sparkfun/SparkFun_ATECCX08a_Arduino_Library #2: Problem with Sign Example on Arduino Zero

boseji, 2020-02-24:
Hello,

I have a spare ATECC508A chip hooked up to a Arduino Zero.
While executing the `Example2_Sign` the output gets stuck in the middle.

```
Successful wakeUp(). I2C connections are good.

Serial Number: 	012370A086A03B82EE
Rev Number: 	00005000
Config Zone: 	Locked
Data/OTP Zone: 	Locked
Data Slot 0: 	Locked

uint8_t message[32] = {
0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F, 
0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18, 0x19, 0x1A, 0x1B, 0x1C, 0x1D, 0x1E, 0x1F
};
```

After this It just stalls.
For configuration I executed the `Example1_Configuration`
Here is that output:

```
Successful wakeUp(). I2C connections are good.

Serial Number: 	012370A086A03B82EE
Rev Number: 	00005000
Config Zone: 	Locked
Data/OTP Zone: 	Locked
Data Slot 0: 	Locked

Would you like to configure your Cryptographic Co-processor with SparkFun Standard settings? (y/n)
***Note, this is PERMANENT and cannot be changed later***
***If you do not want to do this, type an 'n' or unplug now.***
```

Please help resolve this.

Thanking you,

Regards
@boseji

lewispg228, 2020-02-24:
Hi @boseji ,
Sorry you're having trouble using our library. Thanks for reaching out. I have never used it with an Arduino Zero board, and my first guess is that it's an I2C buffer length problem.

If configuration is working, then I'm assuming that most I2C communications are working properly. Do you happen to know what the I2C buffer length is when using that board? It needs to be at least 128 bytes, and so the standard Wire library for most Arduinos is 32 Bytes, and this needs to be modified in the wire.h header file.

For reference, we have verified this library on SamD boards and the SparkFun Artemis (Apollo 3), and these do not require any modification to the buffer length. If you have another Arduino with either of these architectures, that would be a good test (as a point of reference).

Also, what exact model of the Arduino Zero are you using?

Let us know and thanks again for reaching out,
-Pete

boseji, 2020-02-25:
Hi Pete,

Thanks for your reply.

Arduino Zero is the first SAMD board:
<https://store.arduino.cc/usa/arduino-zero>

I am using the Programming USB port on this board with the same variant selected.

Here is the Ring Buffer size being used:
<https://github.com/arduino/ArduinoCore-samd/blob/1.8.4/libraries/Wire/Wire.h#L74>

This is the SAMD official core I am using at release version `1.8.4`.
Since this is an official SAMD and Sparkfun library supports SAMD, so I went for this board.
Unfortunately I don't have an Artemis board to verify.

I am also having a ESP32 board : <https://docs.zerynth.com/latest/official/board.zerynth.doit_esp32/docs/index.html>

If you wish for checking, I can change the following line:
<https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library/blob/master/src/SparkFun_ATECCX08a_Arduino_Library.h#L94>

To use Serial like Apolo3. Then we can have a 1:1 compare.

ESP32 Core release '1.0.4` supports 128 bytes:
<https://github.com/espressif/arduino-esp32/blob/1.0.4/libraries/Wire/src/Wire.h#L34>

Please let me know.

Warm Regards,
@boseji

N5066, 2020-07-31:
Hey!
I have tried to use this library on Arduino IDE to use it with Arduino UNO...But when I compiled the Configuration Example there was an error :

Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\SMDP-C2SD\Documents\Arduino\Example1_Configuration\Example1_Configuration\Example1_Configuration.ino:27:0: warning: "SerialPort" redefined

 #define SerialPort Serial

 ^

C:\Users\SMDP-C2SD\Documents\Arduino\Example1_Configuration\Example1_Configuration\Example1_Configuration.ino:25:0: note: this is the location of the previous definition

 #define SerialPort SerialUSB

 ^

In file included from C:\Users\SMDP-C2SD\Documents\Arduino\Example1_Configuration\Example1_Configuration\Example1_Configuration.ino:29:0:

C:\Users\SMDP-C2SD\Documents\Arduino\libraries\SparkFun_ATECCX08a_Arduino_Library\src/SparkFun_ATECCX08a_Arduino_Library.h:197:108: error: 'SerialUSB' was not declared in this scope

  boolean begin(uint8_t i2caddr = ATECC508A_ADDRESS_DEFAULT, TwoWire &wirePort = Wire, Stream &serialPort = SerialUSB);  // SamD21 boards

                                                                                                            ^

C:\Users\SMDP-C2SD\Documents\Arduino\Example1_Configuration\Example1_Configuration\Example1_Configuration.ino: In function 'void setup()':

Example1_Configuration:37: error: call to 'boolean ATECCX08A::begin(uint8_t, TwoWire&, Stream&)' uses the default argument for parameter 3, which is not yet defined

   if (atecc.begin() == true)

                   ^
call to 'boolean ATECCX08A::begin (uint8_t, TwoWire&, Stream&)' uses the default argument for parameter 3, which is not yet defined

Looking forward to yours reply

lewispg228, 2020-07-31:
Hey @N5066 ,
As you've experienced, this library is not yet supported on the Uno. If you want to attempt to get it working, then you'll need to adjust some of the source files to correctly handle Serial, and also adjust the Wire.h file to have a larger I2C buffer. 

I was actually working on this last week, but wasn't able to get it to work on an Uno. I haven't looked at what's happening with a logic analyzer yet, but hopefully I can find some time next week to dig into this further.

I'll keep you posted, and good luck!

N5066, 2020-08-01:
Thanks for your reply...
Though I will be trying myself yet I will be waiting for your mail when you
will make changes in the library...

m0923678421, 2023-10-12:
It's work on UNO,but MEGA2560 is better.
Because if use debug in library.
`  if (!checkCount(1) || !checkCrc(1))
    return false;`

UNO's memory is not enough to use and it willl not work current.

I meet this problem several years ago with Funduino Colorduino.
https://youtu.be/QuG5agVdF58
I2C buffer size is not enough to package command.

Some libraries need to be change.

1.
wire.h
location : C:\Users\"you pc name"\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4\libraries\Wire\src\Wire.h

`#define BUFFER_LENGTH 32` 

32 change to 96

`#define BUFFER_LENGTH 96`

2.
twi.h
location : C:\Users\BEAR_PC\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4\libraries\Wire\src\utility\twi.h

 `#define TWI_BUFFER_LENGTH 32`

32 change to 96

 `#define TWI_BUFFER_LENGTH 96`

VinayBharambe, 2024-10-21:
@m0923678421 Thanks. It solves the problem. BTW I tried it on Arduino DUE.