sparkfun/SparkFun_ATECCX08a_Arduino_Library #5: It won't compile until changing SamD21 boards serial port

terceranexus6, 2020-06-02:
I had to edit the line
#define SerialPort SerialUSB
to
#define SerialPort Serial

in `SparkFun_ATECCX08a_Arduino_Library.h` _SamD21_ boards configuration in  in order to make it compile. I don't know if this is a common issue but it took a while until I found the solution so I wanted to share.

By the way thank you for the updated examples, they are very useful!

lewispg228, 2021-01-08:
Hi @terceranexus6 , 

Sorry for my delay in responding to this issue (been a crazy year). Thanks for submitting this, but I can't seem to find the line of code you are talking about. I am also testing this with a [Pro RF SAMD21](https://www.sparkfun.com/products/15836) and it seems to be compiling okay for me now. 

There is a check against the architecture being used and that sets the serial port, so maybe this is fixed?

https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library/blob/master/src/SparkFun_ATECCX08a_Arduino_Library.h#L194

What hardware are you using and are you still seeing this compile error?

Let me know and thanks!
-Pete

PaulZC, 2023-01-24:
This is resolved in v1.3.1. Thanks for reporting!

closed 2023-01-24 by PaulZC