sparkfun/SparkFun_ATECCX08a_Arduino_Library #7: Problem with signature verification on an ESP32

rz259, 2020-10-27:
Hello,

i am struggling to get the signature verification done for an ESP32 board. I have tried several commands (sign, Info, SHA256), all of them work without trouble.

However when it comes to verify the signature created with the sign-command, then I get into trouble. I have verified by a self-written Java program that the signature generated by the ATECC608A is correct, it's the verification command that's causing trouble.

First, the delay time of 58 ms is not enough, also the maxRequests of 20 in the method verifySignature is not enough.
I have increased the delay to a second (1000 ms) and finally get a result after 32 iterations in the method verifySignature.
However I get a status code of 0x11 (meaning the chip got a wakeup signal) whereas a 0x00 would be expected.

I have this behaviour on a ATECC508A and could reproduce it on a ATECC608A. Up to now I could not get any signature verified.
Do you have an idea why I get a status code of 0x11?. I'd understand if I got an "execution error" (0x0F), however a 0x11 seems pretty weird to me...

I run the i2c bus with with a frequency of approx. 250000. A lower frequency does not work either and I am unable to increase it any further because the chip won't be recognized at higher speeds.

rz259, 2020-10-28:
I have tried the verification with a Raspberry Pi and the CryptoAuthTools from Microchip and had success in doing so. Thus the chip itself is not defective and there must be either a problem in the library or the ESP32 is not able to do the verification.
However I wonder why other commands like sign oder sha256 work without problems with the ESP32.

lewispg228, 2020-10-28:
Hi there,
Thanks for reaching out. We have not yet tried this on an ESP32 board. My guess is that the I2C buffer is not large enough. Can you verify that it is 256 bytes? (You need a larger I2C buffer to do the verification command because it contains so much: command, message, signature and CRC). All of the other commands you are seeing success with are smaller in length. 

Hope this helps and let us know!
-Pete

rz259, 2020-10-28:
Hello Pete,

thank you for your response. Which buffer are you talking exactly about? Are you talking about the variable inputBuffer?

`byte inputBuffer[BUFFER_SIZE]; // used to store messages received from the IC as they come in`
 (currently line 200 in the headerfile)

If yes, then my answer would also be yes, I set BUFFER_SIZE to 256. Anyway the corresponding method receiveResponseData reads the answer in chunks of only 32 bytes. The code contains a comment that this had been done to avoid an overflow for the atmega32.

Or e are you talking about the internal I2C-buffer?

lewispg228, 2020-10-28:
Yes, the internal I2C buffer. It usually is set somewhere in the wire library.

rz259, 2020-10-28:
Hi Pete,

thanks for your hint. Now it works perfectly!!!!
The buffersize was 128, I have set it to 256 and it worked immediately. I then started playing around and found  out that the buffersize can be as low as 136 but not lower. If the buffersize is less than 136 then the error 0x11 is produced.

Thank you very much for your help!!!

I hope this issue will help someone else trying to use the ATECC608A with an ESP32.

closed 2020-10-28 by rz259

lewispg228, 2020-10-28:
You're welcome. Glad it worked out for you. Cheers to secure authentication!