sparkfun/SparkFun_ATECCX08a_Arduino_Library #8: SAM21pro RF  BOB and Alice Example

camilojimo, 2020-12-15:
Hi I am trying to get the examples to work and I am havving some problems with teh verification process. For some reason I alwas get only half the signature over serial port 

here is the output from the serial port

uint8_t AlicesPublicKey[64] = {
0x2C, 0xB8, 0x1D, 0x95, 0xD4, 0xF2, 0x00, 0x1A, 0x92, 0xC3, 0x1D, 0x55, 0xD8, 0x4D, 0x35, 0xA4, 
0xCF, 0xB8, 0xCD, 0x6B, 0x32, 0x60, 0x69, 0xE5, 0x6B, 0x8C, 0xAC, 0x15, 0xB7, 0x35, 0x9A, 0x86, 
0xD0, 0xC6, 0xF1, 0x8F, 0x87, 0xDC, 0xA0, 0x2F, 0x4C, 0x8E, 0x21, 0x1C, 0x25, 0xDD, 0x2C, 0x9E, 
0xA2, 0x2D, 0x8A, 0xE7, 0xB6, 0xF1, 0x83, 0x7B, 0x0D, 0xC2, 0x21, 0x53, 0x1E, 0x9B, 0xEF, 0x79
};

Hi I'm Bob, I'm listening for incoming messages from Alice on my RX1 pin.

0
0
0
0
0
0
0
0
0
0
24
24
24
Message Received!

uint8_t message[32] = {
0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F, 
0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18, 0x19, 0x1A, 0x1B, 0x1C, 0x1D, 0x1E, 0x1F
};

uint8_t signature[64] = {
0x80, 0xA2, 0x4F, 0x35, 0xA1, 0x4F, 0xB7, 0x97, 0xD4, 0x37, 0x5C, 0xCC, 0x4D, 0x0F, 0x31, 0x5F, 
0x05, 0x57, 0x30, 0xA2, 0xFE, 0x78, 0x11, 0x9F, 0xA2, 0x7B, 0x3D, 0xF1, 0xD0, 0x5F, 0x3E, 0xFF, 
0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 
0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF
};

Can anyone helpme with this issue ?
Thanks

lewispg228, 2020-12-15:
Hi camilojilo,

It may be that your I2C buffer length is set to 32 bytes, and so can't receive the entire signature.

What microcontroller are you using? Most Arduino's that use the Wire library have their I2C buffer length set to 32 bytes. This library is currently only supported to work with the Artemis, which has a larger I2C buffer (256 bytes).

lewispg228, 2021-01-08:
Hey @camilojimo ,

On second thought, I think this may have to do with the serial buffer size, and nothing to do with the I2C buffer size. I don't have two Pro RFs to try this out on, but it'd probably be worth a shot. Try increasing the SERIAL_BUFF_SIZE from 64 up to 256.

As I saw in this [forum post](https://forum.arduino.cc/index.php?topic=241916.0), I think you will need to adjust the SERIAL_BUFFER_SIZE in the RingBuffer.h file, which is found in the core files here:

[image]

As a side note, this problem may not have to do with the serial buffer size (because it looks like the Artemis also has a serial buffer size set to 64), but it might be something to do with timing and how fast the artemis can pull it it vs the SamD boards. This is what I am currently seeing for the apollo3 serial buffer size:

[image]

Either way, I think it's worth a shot on the SamD21s. Please try this out and let us know how it goes.

Cheers!
-Pete