sparkfun/SparkFun_ATECCX08a_Arduino_Library #13: Arduino WAN 1310 signing issue

perone, 2022-01-11:
Hi, I did the initial config and locking on an Arduino WAN 1310 and everything went well, except that it is not able to sign a message, is this something known ? I guess it could be related to the [key type](https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library/blob/master/src/SparkFun_ATECCX08a_Arduino_Library.cpp#L1062) but I'm not sure what other reasons it could be, the `loadTempKey` is returning true but the `signTempKey` is returning false for the slot 0. Any help would be appreciated.

perone, 2022-01-12:
I found the issue, it turns out that it was because of the delay for the signing op, it was always taking more time than specified in the datasheet, I'll submit a PR.

lewispg228, 2022-01-12:
Pull request #14 merged. Closing this issue. Thanks again @perone !

closed 2022-01-12 by lewispg228