sparkfun/SparkFun_ATECCX08a_Arduino_Library #16: Problem with the sign verification example number 3

santcorc, 2022-03-28:
Hello, I am using an ESP32 WROOM-32 NodeMCU v2 with the cryptographic chip ATECC508A and when trying to verify the sign from the second example it always returns a verification failure. I followed the steps carefully so I do not know what is wrong here. Below are the outputs given by the second and third example.

Example 2 output 

This device's Public Key:

uint8_t publicKey[64] = {
0xB2, 0xAA, 0xE7, 0x84, 0x1D, 0x43, 0x5C, 0xE6, 0x49, 0xFD, 0x26, 0x3B, 0x8D, 0xC2, 0xF8, 0x2A, 
0x20, 0x49, 0x9A, 0xFC, 0xAE, 0xFE, 0x25, 0x1C, 0x6A, 0x90, 0x26, 0xC6, 0x40, 0xC3, 0x4C, 0x5F, 
0x3A, 0x98, 0xAA, 0xA4, 0x2B, 0xFE, 0x46, 0x40, 0x99, 0xB4, 0xC5, 0x26, 0x81, 0x94, 0x6B, 0x18, 
0xDF, 0x3D, 0xE6, 0x18, 0x6D, 0x4C, 0x61, 0xE0, 0x1F, 0xD2, 0x4F, 0x73, 0x80, 0xB2, 0x2E, 0x68
};

uint8_t message[32] = {
0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F, 
0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18, 0x19, 0x1A, 0x1B, 0x1C, 0x1D, 0x1E, 0x1F
};

uint8_t signature[64] = {
0x63, 0x5D, 0xFD, 0x43, 0x0C, 0x05, 0xE3, 0x19, 0xE2, 0x1C, 0x1E, 0xC8, 0x9E, 0xD2, 0x49, 0xCC, 
0x91, 0x54, 0x55, 0xB0, 0xC2, 0x4C, 0xDE, 0x74, 0xE9, 0x75, 0x36, 0xE3, 0xE6, 0x19, 0xB6, 0x89, 
0x9D, 0xBD, 0x4D, 0x84, 0x3F, 0xDB, 0x2B, 0xE0, 0xA6, 0xA4, 0x30, 0xAB, 0x24, 0x03, 0xE3, 0x84, 
0xB3, 0xCD, 0xDD, 0xD3, 0x1D, 0x03, 0x29, 0x0C, 0x4B, 0xF0, 0x4C, 0xE7, 0x48, 0x17, 0x2B, 0x40
};

Example 3 output

Successful wakeUp(). I2C connections are good.

Serial Number: 	0123C5C7349171DDEE
Rev Number: 	00005000
Config Zone: 	Locked
Data/OTP Zone: 	Locked
Data Slot 0: 	Locked

uint8_t publicKeyExternal[64] = {
0xB2, 0xAA, 0xE7, 0x84, 0x1D, 0x43, 0x5C, 0xE6, 0x49, 0xFD, 0x26, 0x3B, 0x8D, 0xC2, 0xF8, 0x2A, 
0x20, 0x49, 0x9A, 0xFC, 0xAE, 0xFE, 0x25, 0x1C, 0x6A, 0x90, 0x26, 0xC6, 0x40, 0xC3, 0x4C, 0x5F, 
0x3A, 0x98, 0xAA, 0xA4, 0x2B, 0xFE, 0x46, 0x40, 0x99, 0xB4, 0xC5, 0x26, 0x81, 0x94, 0x6B, 0x18, 
0xDF, 0x3D, 0xE6, 0x18, 0x6D, 0x4C, 0x61, 0xE0, 0x1F, 0xD2, 0x4F, 0x73, 0x80, 0xB2, 0x2E, 0x68
};

uint8_t message[32] = {
0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F, 
0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18, 0x19, 0x1A, 0x1B, 0x1C, 0x1D, 0x1E, 0x1F
};

uint8_t signature[64] = {
0x63, 0x5D, 0xFD, 0x43, 0x0C, 0x05, 0xE3, 0x19, 0xE2, 0x1C, 0x1E, 0xC8, 0x9E, 0xD2, 0x49, 0xCC, 
0x91, 0x54, 0x55, 0xB0, 0xC2, 0x4C, 0xDE, 0x74, 0xE9, 0x75, 0x36, 0xE3, 0xE6, 0x19, 0xB6, 0x89, 
0x9D, 0xBD, 0x4D, 0x84, 0x3F, 0xDB, 0x2B, 0xE0, 0xA6, 0xA4, 0x30, 0xAB, 0x24, 0x03, 0xE3, 0x84, 
0xB3, 0xCD, 0xDD, 0xD3, 0x1D, 0x03, 0x29, 0x0C, 0x4B, 0xF0, 0x4C, 0xE7, 0x48, 0x17, 0x2B, 0x40
};

Verification failure.

Thank you in advance

themagicm, 2023-07-31:
same problem as I have.   You didnt get an answer...hopefully I get one and it helps you.

lewispg228, 2023-07-31:
Hi there @santcorc , Sorry this issue has sat for so long. I hope you were able to get it working on your own, but if you are still having issues, it may be worth trying to increase your I2C buffer size to 256. This worked for @themagicm on their recent similar issue. Let us know and again, sorry for the delay here. -Pete

https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library/issues/22