sparkfun/SparkFun_ATECCX08a_Arduino_Library #18: What is writeConfigSparkFun

alberk8, 2022-10-22:
Can someone explain to me what this does and where is the document. 

https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library/blob/155ccbc24f17be812a935d12c2f024f9abab866f/src/SparkFun_ATECCX08a_Arduino_Library.cpp#L1054

lewispg228, 2022-10-24:
Hi @alberk8 ,
There is a high-level description of what the function does just above the function, here:

https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library/blob/155ccbc24f17be812a935d12c2f024f9abab866f/src/SparkFun_ATECCX08a_Arduino_Library.cpp#L1048

Each key slot on this device must be configured properly to allow certain features. Both the keyType and the slotConfig for each slot must be programmed before the config is locked. We are setting up the slot to work as we want for our examples to work. You can read more about all the available settings for each key slot on page 20 of the datasheet here:

https://github.com/sparkfun/SparkFun_Cryptographic_Co-Processor_Breakout_ATECC508A_Qwiic/blob/master/Documentation/datasheet-ATECC508A.pdf

SlotConfig Bits for slots begin at byte 20 of the EEPROM (aka part of the config zone). KeyConfig Bits for slots begin at byte 96 of the EEPROM.

Each bit within each settings byte all work together to get the desired functionality to work. Unfortunately, that leads to quite a bit of jumping around the datasheet and double-checking. Also, you can't really test your settings until the config is locked, so that means if you get them incorrect, then that IC is no longer functional. So, unfortunately, until you gain lots of experience with this IC, it means that it is fairly difficult to avoid bricking at least one or two ICs during new configuration experimenting.

If you are interested in setting up the IC to do other features outside of our examples, then I'd like to suggest you reach out to the forum to see if there are any other users doing something similar.

Hope this helps,
Pete

alberk8, 2022-10-25:
Thank you for the reply.  What I don't understand for example is SlotConfig, where the (20 is divided by 4)  `result2 = write(ZONE_CONFIG, (20 / 4), data2, 4);`

lewispg228, 2022-10-26:
There is a lot of information about address encoding on page 58 of the datasheet (section 9.1.4). But basically, the slots/blocks/word-offsets are designed in a way that makes it efficient to create the actual word address you need to send. (Specifically, the fact that data is accessed on a 4 byte word boundary). Without going into all the actual steps needed to create the necessary param2 address, you can simply take the desired address and drop the least significant two bits. (This can be accomplished by dividing by 4). 

[image]

Note, address encoding is different for the config/otp zones vs the data zone. Because we are working within the config zone for these writes, we are able to divide by four and the address byte is "scooted" over (dropping least sig two bits) to indicate the necessary block and offset:

[image]

Byte 20 is B0001 0100.

divide by 4 and you get:

B0000 0101

Bits [4:3] are our block: 0
Bits [2:0] are our offset: 5

Byte 96 is B0110 0000

divide by 4 and you get:

B 0001 1000

Bits [4:3] are our block: 3
Bits [2:0] are our offset: 0

Hope this helps!
Pete

alberk8, 2022-10-27:
Thank you.  This is a fantastic and clear explanation.

alberk8, 2022-10-27:
The specs says writing/reading is 4 or 32 bytes.  when writing to the 4 bytes operation (slot config).  which of the 2 bytes are of significant (the 2 MSB or 2 LSB) ?.