sparkfun/SparkFun_ATECCX08a_Arduino_Library #22: Problem reading using this lib on esp32 + ECC508A

themagicm, 2023-07-31:
I'm using an ESP32-WROOM-32D connected to an ECC508A via SDA and SCL. Both have 4.7k resistors.

Libraries used:

```
#include <ArduinoECCX08.h>
#include <utility/ECCX08SelfSignedCert.h>
#include <utility/ECCX08DefaultTLSConfig.h>
#include <EEPROM.h>
#include "CRC32.h"
```

on the ESP32 EEPROM i write the input, signature and public key.

On the ECC508A, slot 0, I store the private key.

Everything writes just fine to the esp32 eeprom and I'm assuming the ECC508A. I can dump the input, sig and pub key from EEPROM on the esp32 fine. Then I get this error below when I run :

`ECCX08.ecdsaVerify(settings.input, settings.signature, settings.public_key)`

Error is:
```

    [ 2188][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263

    [ 4189][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263 ..... etc etc.
```

It supposedly does communicate with the the ECC508A:
```

    Attempting to communicate with ECC508/ECC608...
    Successfully established communication with ECC508/ECC608.
```

I've tried to add Wire.settimeout(2000); to see if it helps but it doesnt. I'm totally stumped.

lewispg228, 2023-07-31:
Hi there,
Sorry you've been having issues with this example.

I am currently rounding up some hardware to see if I can replicate this issue.

For now, I do have one idea for you to try:

These examples do require a larger-ish i2c buffer. If you can adjust your wire.h file before compiling, I might try increasing this from 128 to 256. I believe the hookup guide was created with the Artemis because this defaults to 256.

For reference, it is located here in the espressif github repo: 

https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L43

And it usually lives deep within your arduino install here-ish:

C:\Users\user.name\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\Wire\src

But I'm also wondering if something changed with the recent big update to Arduino "2.0", so we may need to dig deeper.

I'll keep ya posted on how testing goes on our end.
Thanks,
-Pete

themagicm, 2023-07-31:
PETE!!! YOU GENIUS!

Wire.setBufferSize(256);
WORKED!

Sent with [Proton Mail](https://proton.me/) secure email.

lewispg228, 2023-07-31:
Excellent! Glad it was resolved so easily and qwiicly for you :) Hope the rest of your authentication project goes well! This reminds me, I need to update my garage door opener to accept new public keys.

themagicm, 2023-07-31:
cool.  whats the private key?  lol I kid I kid.